Devdevdavid/Pascal-Project #3: [Interface température] Utilisation du port USB lorsque l'ESP est connecté à la carte

Devdevdavid, 2020-08-08:
# Symptome

Un problème empêche l'USB de fonctionner correctement lorsque le module WEMOS est connecté à la carte électronique.
Le PC ne reconnait pas le périphérique lorsque le module WEMOS est connecté à la carte.

# Hypothèses

- Une pin de l'USB est utilisé par la carte électronique
- Une pin est asservie au moment du boot de l'ESP (Wake-up, reset, sleep mode...) Voir [Pinout de l'ESP8266](https://randomnerdtutorials.com/esp8266-pinout-reference-gpios/)
- Un problème d'alimentation (Le régulateur du module ne suffit pas à alimenter toute la carte électronique

Pascal-Tours, 2020-08-08:
Coucou,

 

Merci pour le retour,

J’ai trouvé le problème, à savoir un retour du 5v qui revient sur le 3V3…

Je fais mes modifs et je te tiens au courant.

@+

Devdevdavid, 2020-08-14:
As-tu une nouvelle révision du **schéma de la carte** à mettre sur GitHub ?

Pascal-Tours, 2020-08-14:
Coucou,

 

T’es fort ! J

Oui je te la joint, je câble le platine relais.

@+

Pascal

Devdevdavid, 2020-08-17:
Voir le schéma annoté [board_temp_domoticz-telegram_relais_Rev.B-Annoté.pdf](https://github.com/Devdevdavid/Pascal-Project/blob/add_telegram_interface/interface_temperature/electronique/schemas/board_temp_domoticz-telegram_relais_Rev.B-Annoté.pdf) pour la correction de la confusion entre 3V3 et 5V.

closed 2020-08-17 by Devdevdavid