Devdevdavid/Pascal-Project #6: [Interface température] Ajouter un mode de comparaison par delta

Devdevdavid, 2020-08-08:
Il serait intéressant de pouvoir comparer la valeur des deux sondes de température entre elles.
Ceci serait notamment utile pour détecter s'il est viable d'ouvrir la fenêtre lorsqu'il fait froid/chaud.

Devdevdavid, 2020-08-09:
J'ai ajouté la fonctionnalité de comparaison différentielle. 

* L'algorithme effectue la différence entre le capteur Temp. 1 et Temp. 0 et prend la valeur absolue (aka `abs(T1 - T0)`).
* C'est cette valeur qui est comparée au seuil `SCRIPT_TEMP_ALERT_DIFF_THRESHOLD` pour déterminer si l'on est en alerte ou non.
* _Note_ : La valeur d'hysteresis `SCRIPT_TEMP_ALERT_HYSTERESIS` est toujours utilisée avec cette méthode de comparaison.
* Le mode choisi (`METHOD_THRESHOLD` ou `METHOD_DIFFERENTIAL`) est indiqué dans la commande `/sensors`de l'interface Telegram.

Commit a3142ee81ea279060ab0c2d856773b99db12f100

Pascal-Tours, 2020-08-12:
OK

closed 2020-08-12 by Pascal-Tours