Devdevdavid/Pascal-Project #8: [Interface température] Affichage des seuils d'alertes configurés

Pascal-Tours, 2020-08-12:
Affichage des seuils d'alertes configurés dans le global.h.
`SCRIPT_TEMP_ALERT_SENSOR_0` et `SCRIPT_TEMP_ALERT_SENSOR_1`

Devdevdavid, 2020-08-14:
Affichage des seuils configurés dans le retour de la commande `/sensors` en fonction de la méthode de comparaison choisie :
* `METHOD_THRESHOLD` : Affichage de `SCRIPT_TEMP_ALERT_SENSOR_0` et `SCRIPT_TEMP_ALERT_SENSOR_1`
> Les capteurs sont en mode SEUIL
> Temp. 0] Capteur non utilisé (Seuil configuré: 28.00°C)
> Temp. 1] Capteur non utilisé (Seuil configuré: 35.00°C)
* `METHOD_THRESHOLD` : Affichage de `SCRIPT_TEMP_ALERT_DIFF_THRESHOLD` 
> Les capteurs sont en mode DIFFERENTIEL (T1 - T0 >= 4.00°C)
> Temp. 0] Capteur non utilisé
> Temp. 1] Capteur non utilisé

Commit 39143dc22cd63927762e5306b4e77dec4be1cfa5

Pascal-Tours, 2020-08-19:
Affichage OK

closed 2020-08-19 by Pascal-Tours