Devdevdavid/Pascal-Project #15: [Interface Température] Ajout d'un serveur telnet

Devdevdavid, 2020-12-15:
La mise à jour par OTA c'est top mais quand l'ESP est dans son boitier à 3m de hauteur, plus moyen d'utiliser la liaison série pour débogguer et continuer ses développements.

Je propose donc d'ajouter un serveur telnet afin de pouvoir consulter les logs à partir d'un client Telnet depuis son PC à distance.

L'idée est de connecter le serveur Telnet au module de communication série déjà présent.

Un exemple d'implémentation ici : [WiFiTelnetToSerial.ino](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/examples/WiFiTelnetToSerial/WiFiTelnetToSerial.ino)

bastian-bou, 2020-12-15:
Oui c'est une bonne idée ! le telnet n'est pas sécu mais de toute façon tu es connecté sur la box donc réseau local.
Je suis d'accord

Devdevdavid, 2020-12-19:
Serveur Telnet ajouté (7b5cc5872907516d7617b466345c96662851a175) avec refonte de l'interface `serial` pour laisser apparaître le module `term`.

closed 2020-12-28 by Devdevdavid