Devdevdavid/Pascal-Project #16: [Interface Température] SPIFFS est déprécié

Devdevdavid, 2020-12-16:
Le module FILE_SYS (file_sys.cpp) utilise le système de fichier SPIFFS pour fonctionner.

Or, à la compilation, PlatformIO indique que cette librairie ne devrait plus être utilisée.

Un équivalent est proposé : LittleFS

bastian-bou, 2020-12-16:
Let's Gooooooo

GilDev, 2020-12-16:
Leeeeeeeeeeeeeeet’s goooooooooo!!

Devdevdavid, 2021-01-09:
La dépréciation de SPIFFS n'est valable que pour l'ESP8266 car il n'y pas encore à ce jour de portage valable pour LittleFS sur ESP32. (J'ai testé [lorol/LITTLEFS](https://github.com/lorol/LITTLEFS) sans succès)

J'en ai donc profité pour séparer clairement les environnements de travail sur les deux plateformes en créant `base_esp32`et `base_esp8266` dans le fichier `PlatformIO.ini`

Aussi, le webserveur repose désormais sur le module `file_sys` (Qui se trouve ne pas être optionnel).

Nous utilisons désormais LittleFS pour les ESP8266, et SPIFFS pour les ESP32. C'est la variable `G_FileSystem` qui permet de différencier les deux.

Les modifications sont disponibles sur le commit a6d332405aa443a978bc9bbae0d9538a13bcb5c8.

> Note 1 : A ce jour, nous n'utilisons pas le système de fichier lorsque nous sommes sur une plateforme ESP8266. De ce fait, nous ne pouvons pas valider le fonctionnement de `LittleFS` avec l'existant.

> Note 2 : les systèmes de fichier LittleFS et SPIFFS ne sont pas compatibles. LittleFS  est plus rapide que SPIFFS.

closed 2021-01-09 by Devdevdavid