Devdevdavid/Pascal-Project #17: [Interface Température] Configuration des paramètres wifi par l'utilisateur

Devdevdavid, 2020-12-28:
Dans le but de rendre l'utilisation de LightKit plus accessible à l'utilisateur, on souhaite que les configurations wifi (SSID, mot de passe, IP, Gateway, Masque) soient configurables.

La procédure serait la suivante :
- a-t-on une configuration valide enregistrée en mémoire flash ?
- si Oui:
 - on essaie de se connecter au réseau indiqué dans la configuration
- si Non:
 - On démarre en mode Access Point
 - L'utilisateur peut se connecter au module à travers l'interface web (IP par défaut à définir)
 - Il peut choisir de configurer le module pour se connecter à un réseau wifi existant
 - Il saisi les paramètres dans un formulaire HTML
 - Note: il serait intéressant d'afficher les réseaux alentours détectés avec une indication de la force du signal
 - Après validation du formulaire, les paramètres sont stockés en mémoire flash
 - Une séquence de reboot est enclenchée

Les objectifs secondaires:
1. La configuration lue au moment du boot doit être affichée dans le terminal
2. On utilisera un checksum ou un CRC pour valider la configuration flash
3. La sauvegarde des paramètres doit être évolutive et doit supporter l'ajout de nouveaux paramètres
4. La led de status, s'il y en a, doit indiquer si le module est en mode connecté ou AP

D'autres propositions ?

Devdevdavid, 2021-01-15:
Sur l'ESP32, j'obtiens désormais des crashs en mode AP lors d'une tentative de connexion. Ce problème est évoqué [ici](https://github.com/espressif/arduino-esp32/issues/863#issuecomment-443539477)

Devdevdavid, 2021-01-18:
Le problème venait d'un malheureux `memcpy()` qui avait pour destination une adresse de pointeur plutôt qu'un pointeur.
Résolu avec 8a814e8d0a1b5950415b229b27a86c1675896eec

Devdevdavid, 2021-01-18:
📸 Voici un aperçu de la page de configuration Wifi :
[Lightkit.pdf](https://github.com/Devdevdavid/Pascal-Project/files/5832425/Lightkit.pdf)
[Lightkit 2.pdf](https://github.com/Devdevdavid/Pascal-Project/files/5832426/Lightkit.2.pdf)

GilDev, 2021-01-19:
Pas mal ! Ça ressemble à Tasmota, peut-être la source d’inspiration. 🙂

Devdevdavid, 2021-01-19:
Non je ne connaissais pas ^^

Devdevdavid, 2021-01-20:
Voici une première ébauche de l'interface des réseaux wifi découverts avec le SSID et RSSI affiché sous forme de logo :
[image]

Devdevdavid, 2021-01-24:
Scan des réseaux alentours fonctionnel. Il suffit de cliquer sur un scan pour pré-remplir le champ SSID.
Lors du scan, le module se déconnecte du réseau, lance un scan, puis se reconnecte. Idem lorsqu'il est en mode AP.
Lors du scan, il n'est donc plus possible d'interagir avec le module. Cela crée des latences si l'on clique sur un bouton de la page web en même temps.
Le scan est relativement rapide (3-5s) donc ça passe.
On prend le soin d'enregistrer en RAM le canal wifi et l'adresse MAC de la box avant la déconnexion pour une re-connexion plus rapide.

Devdevdavid, 2021-01-27:
# Déconnexion de l'utilisateur lors du scan

Lors des tests avec Pascal, nous avons remarqué que les téléphones Android se déconnectent du module en point d'accès lors du scan wifi des réseaux environnants.
Cette déconnexion empêche la validation du formulaire HTML et donc la configuration du module.

Pour éviter cela, nous avons choisi d'effectuer un scan peu après le démarrage et de conserver les résultats pour 24h.
Au delà, le chargement de la page web impliquera une demande de rafraichissement des scans qui déconnectera les utilisateurs Android. Pas de soucis pour leur 2nd tentative car les résultats seront de nouveau disponibles pour 24h.

En pratique, l'utilisateur va brancher son module puis tenter une configuration wifi. Ce problème ne sera apparent que pour un faible nombre d'utilisateur et n'est pas bloquant.

> Note: Pas de problème pour un module en mode client, l'utilisateur reste connecté à sa box

Commit : dcb2b8313c1d36dd5e3f334903decc24e4c74b55

Devdevdavid, 2021-02-02:
Tester le point sur la LED de statut avec la carte board_temp_telegram_relay qui dispose d'une led de status de type Néopixel

Pascal-Tours, 2021-02-03:
Fonction validée !

closed 2021-02-03 by Devdevdavid