Devdevdavid/Pascal-Project #19: [Interface Température] Utilisation de mDNS pour éviter de connaître l'IP du module pour s'y connecter

Devdevdavid, 2021-01-09:
[Cet article](https://techtutorialsx.com/2020/04/17/esp32-mdns-address-resolution/) montre comment utiliser mDNS pour effectuer une résolution de nom de domaine en local sur le module wifi.

Le nom de domaine à utiliser doit être unique, il faut donc qu'il soit configurable par l'utilisateur.

Ce même nom de domaine devrait s'afficher à côté de l'IP au moment du boot dans la liaison série.

Plus d'info sur le protocole mDNS ici : [RFC 6762](https://tools.ietf.org/html/rfc6762)

Devdevdavid, 2021-01-10:
Il semblerait que la librairie `ArduinoOTA` gère déjà cette fonctionnalité. Peut-être qu'une simple configuration suffirait.

GilDev, 2021-01-14:
Je confirme, j’ai déjà réussi à faire fonctionner mDNS avec ArduinoOTA sans souci par le passé.

Devdevdavid, 2021-02-04:
✅ Fonction implémentée avec le commit 07693a6688ec770c61f7bb8ac149c87516d0e031

Le nom du module est configurable depuis la page du wifi.
On accède au module en tapant `<nom>.local` dans son navigateur.
Si on configure un nom vide, alors on n'utilise pas le mDNS.

__Fun fact__ : `Bonjour` c'est l'autre nom de mDNS donné par Apple 🤯

Devdevdavid, 2021-02-04:
Dans mon cas, la résolution de nom avec un ESP8266 ne fonctionne pas sur MacOS. Pas de soucis en revanche avec les ESP32.

Cette différence est expliquée ici : https://github.com/esp8266/Arduino/issues/4790#issuecomment-395207417

Devdevdavid, 2021-02-10:
Suite à quelques tests, nous avons remarqué que la résolution mDNS ne fonctionnait pas sur un PC Windows 7 et un téléphone Android.

mDNS n'est en effet pas supporté nativement par ces plateformes contrairement aux appareils IOS et MacOS.

Sous windows 7, un hack est possible d'après [ce site](https://superuser.com/a/1330028) en utilisant "bonjour" d'Apple
Windows 10 supporte nativement mDNS. Ceci reste à tester.
Pour Android, cela s'annonce mal d'après [ce site](https://android.stackexchange.com/a/51350). Seules quelques version récente d'Android seraient compatibles nativement.

GilDev, 2021-02-11:
En effet j’avais rencontré le même problème. Windows 7 est relativement obsolète maintenant mais Bonjour ça fonctionnait bien. Pour Android c’est une impasse, j’ai cherché un moment mais pas de solution… Si ce n’est pas supporté nativement c’est compliqué.

Devdevdavid, 2021-03-25:
✅ mDNS fonctionnel pour ESP32 et ESP8266 avec commit 58fc617bf178745ef61d4891b34792cb7aaea99f

Devdevdavid, 2021-03-27:
Demande de Pascal : Un mode d'emploi de connexion pour les utilisateurs

commit 22fc0ed4 (2021-03-27): Ajout documentation "Première connexion" closes #19
closed 2021-03-27 by Devdevdavid