Devdevdavid/Pascal-Project #22: [FIRE] Brainstorming sur la board_fire

Devdevdavid, 2021-02-06:
# Contexte
Je souhaite piloter un feu rouge en 230V. J'ai 1 phase par ampoule (C'est des LED rassurez-vous) et 1 neutre commun.

# Problème
J'ai besoin d'une carte de commande pour piloter ces feux. Les 3 ampoules doivent être pilotable individuellement.
J'aimerai utiliser LightKit sur un module wifi pour pouvoir commander le feu à distance.

# Proposition
Vous trouverez ci-dessous un diagramme préliminaire de cette future carte de commande nommée `board_fire`.
Si vous avez des suggestions/améliorations/critiques, n'hésitez pas !

[image]

GilDev, 2021-02-07:
Pourquoi conversion 12V puis 5V et pas 5V direct ?
À quoi sert la sortie OUT 4 ?

bastian-bou, 2021-02-08:
Je te conseille plutôt des opto triac ! comme ça pas besoin de relais, tu commande juste la petite LED de l'opto, et le triac s'occupe du 230VAC ! et le tour est joué !

Pascal-Tours, 2021-02-08:
Regarde le MOC3040 opto triac avec détection du zéro
https://fr.rs-online.com/web/p/optocoupleurs/1611019/

Devdevdavid, 2021-02-10:
> Pourquoi conversion 12V puis 5V et pas 5V direct ?
> À quoi sert la sortie OUT 4 ?

Je m'étais dis que des relais à bobine 12V était plus courant et peut-être moins chers. (Et j'en ai un en stock)
Et puis j'aimerai bien avoir une sortie 12V pour des usages futurs.
Idem pour le relais, pour un éventuel usage futur.

> Je te conseille plutôt des opto triac ! comme ça pas besoin de relais, tu commande juste la petite LED de l'opto, et le triac s'occupe du 230VAC ! et le tour est joué !

C'est formidable ces Opto-Triac !!
Merci messieurs !